HKUST-Aerial-Robotics/Nxt-FC #11: Jlink debug issue

shmm91, 2023-12-18:
Can't connect device using jlink, the nrst/swdio/sclk/gnd gpio is connected, but can't connected with jlink j-flash. The prompt:
Connecting ...
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: J-Link V9 compiled May  7 2021 16:26:12
 - Probe/ Programmer S/N: 59300616
 - Device "STM32H743VI" selected.
 - Target interface speed: 20 kHz (Fixed)
 - VTarget = 3.290V
 - ConfigTargetSettings() start
 - ConfigTargetSettings() end - Took 21us
 - InitTarget() start
 - SWD selected. Executing JTAG -> SWD switching sequence.
 - DAP initialized successfully.
 - InitTarget() end - Took 348ms
 - Found SW-DP with ID 0x6BA02477
 - DPIDR: 0x6BA02477
 - CoreSight SoC-400 or earlier
 - Scanning AP map to find all available APs
 - AP[3]: Stopped AP scan as end of AP map has been reached
 - AP[0]: AHB-AP (IDR: 0x84770001)
 - AP[1]: AHB-AP (IDR: 0x84770001)
 - AP[2]: APB-AP (IDR: 0x54770002)
 - Iterating through AP map to find AHB-AP to use
 - AP[0]: Core found
 - AP[0]: AHB-AP ROM base: 0xE00FE000
 - CPUID register: 0x411FC271. Implementer code: 0x41 (ARM)
 - Cache: L1 I/D-cache present
 - Found Cortex-M7 r1p1, Little endian.
 - FPUnit: 8 code (BP) slots and 0 literal slots
 - CoreSight components:
 - ROMTbl[0] @ E00FE000
 - [0][0]: E00FF000 CID B105100D PID 000BB4C7 ROM Table
 - ROMTbl[1] @ E00FF000
 - [1][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
 - [1][1]: E0001000 CID B105E00D PID 000BB002 DWT
 - [1][2]: E0002000 CID B105E00D PID 000BB00E FPB-M7
 - [1][3]: E0000000 CID B105E00D PID 000BB001 ITM
 - [0][1]: E0041000 CID B105900D PID 001BB975 ETM-M7
 - [0][2]: E0043000 CID B105900D PID 004BB906 CTI
 - I-Cache L1: 16 KB, 256 Sets, 32 Bytes/Line, 2-Way
 - D-Cache L1: 16 KB, 128 Sets, 32 Bytes/Line, 4-Way
 - Executing init sequence ...
 - CPU could not be halted
 - OnDisconnectTarget() start
 - OnDisconnectTarget() end - Took 264ms
 - ERROR: Failed to connect.
Could not perform custom init sequence.
 - ERROR: Connect failed
 Do you have any ideas?

Peize-Liu, 2023-12-18:
I only tested DFU mode. It seems to be the problem with JLINK. Can you provide more information about your debugger and connections? Like JLINK, via OpenOCD, or something else? I might have some idea after that.

shmm91, 2023-12-19:
more info picture
[image]