HKUST-Aerial-Robotics/Nxt-FC #19: PX4 1.14设置外部定位源

Feng1909, 2024-11-05:
请问默认烧录的版本是1.14吗，如何使用EKF2_EV_CTRL设置外部定位源呢，我将这个值设为11，每次接入/mavros/vision_pose/pose的时候，/mavros/local_position/pose的xyz值都会爆炸到万亿级别然后慢慢收敛。另外，在不输入/mavros/vision_pose/pose的情况下/mavros/local_position/pose也会有输出，这与其他一般飞控的使用似乎不太一样，请问这是为什么呢

chenfch3, 2025-04-19:
> 请问默认烧录的版本是1.14吗，如何使用EKF2_EV_CTRL设置外部定位源呢，我将这个值设为11，每次接入/mavros/vision_pose/pose的时候，/mavros/local_position/pose的xyz值都会爆炸到万亿级别然后慢慢收敛。另外，在不输入/mavros/vision_pose/pose的情况下/mavros/local_position/pose也会有输出，这与其他一般飞控的使用似乎不太一样，请问这是为什么呢

请问你最后是怎么解决的

hcws, 2025-05-09:
And I set `EKF2_EV_CTRL = 11` , In position mode , the uav will crack to the ground

Ethan-ZZL, 2026-09-20:
我烧录的是PX4 1.14.3dev，用Mid-360s激光雷达定位没有出现该问题，会不会是外部定位来源的问题？